makerbase-mks/Klipper-for-MKS-Boards #1: MKS TS35 with Klipper

FabioMC, 2021-07-13:
Please, include MKS TS35 into LCD config.

mks-viva, 2021-07-15:
Hi! Klipper currently does not support MKS TS3 screens, this may require modification of the source code.

edouardkleinhans, 2021-09-08:
@mks-viva Maybe we can do a PR to klipper ?

themacboy, 2021-09-18:
Any new about MKS TS25 support on klipper?

akyrassan, 2021-11-22:
Hello,
would you be able to share the wiring diagram?
thanks in advance

themacboy, 2021-11-22:
> > @mks-viva any new about MKS TS35 v2 support on klipper?
> > If yes I will rescue one to use there
> 
> You mean something like that?
> 
> Only figuring out how to calibrate the screen
> 
> [image]

Interesting ! Promising !

willngton, 2021-11-22:
> Hello, would you be able to share the wiring diagram as well as the configuration file? thanks in advance

I have put together a draft on my finding and configuration, its fully operation including touch.

A friend create an easy on time scrip to install, considering you have already installed klipper, klipperscreen via:
https://github.com/th33xitus/kiauh

[MKS_TS35_PI_TELA_MKS.zip](https://github.com/makerbase-mks/Klipper-for-MKS-Boards/files/7584626/MKS_TS35_PI_TELA_MKS.zip)

[TS35_Rascunho_Klipper_RaspberryPI_ENv2.pdf](https://github.com/makerbase-mks/Klipper-for-MKS-Boards/files/7584627/TS35_Rascunho_Klipper_RaspberryPI_ENv2.pdf)

willngton, 2021-11-22:
Video:

https://user-images.githubusercontent.com/5961905/142941528-6624a35b-c8c2-4f95-afc9-c99dc545c9e4.mp4

willngton, 2021-11-23:
https://github.com/willngton/3DPrinterConfig/tree/main/mks_ts35

akyrassan, 2021-11-23:
Thanks ,i will try it 👍

mks-viva, 2022-09-02:
MKS Pi+MKS PI-TS35 and this way is the best.

closed 2022-09-02 by mks-viva

xojikia, 2022-11-10:
Hello, I did everything according to your instructions , The screen is working, it shows everything, but the touch screen is not working
MKS TS35

What is my problem?, can you help me?
Klipper / Mainsail
[image]