makerbase-mks/Klipper-for-MKS-Boards #5: MKS Lite 1.1 can't work

wudixiaobinbin, 2021-08-05:
Mother board is MKS Lite V1.1, step motor can't work, only shaking.

mks-viva, 2021-08-06:
I test it ok: https://github.com/makerbase-mks/Klipper-for-MKS-Boards/tree/main/MKS%20Robin%20Lite%20V1.1
May be you need check your config or motor...

wudixiaobinbin, 2021-08-06:
I test not ok. I can't find the python file, I want to make a bin file myself, may be the klipper version is different from us

wudixiaobinbin, 2021-08-06:
motor is normaly worked with https://github.com/makerbase-mks/MKS-Robin/tree/master/MKS%20Robin%20Lite

mks-viva, 2021-08-06:
So, you can use it. I will close this issues.

closed 2021-08-06 by mks-viva

wudixiaobinbin, 2021-08-06:
hey, my issue is klipper can't work on my MKS lite 1.1

mks-viva, 2021-08-06:
[image]
?

wudixiaobinbin, 2021-08-06:
you tell me to check my motor, I just tell you my motor is worked well, but can't work with klipper

mks-viva, 2021-08-06:
You download my link firmware test?

wudixiaobinbin, 2021-08-06:
Yes I used that firmware, And motor only shaking in place, can't move

mks-viva, 2021-08-06:
Lite V1.1 uses pins to control the reference voltage, but I don’t see the Vref setting in the klipper firmware. You can remove the motor load and test it, if it is will move, so, maybe the current is too small.

mks-viva, 2021-08-06:
https://github.com/makerbase-mks/Klipper-for-MKS-Boards/issues/4

wudixiaobinbin, 2021-08-06:
Aha, Maybe that's the point, Z motor only can go down with heat bed

mks-viva, 2021-08-06:
Yes, but I have not seen klipper's vref settings, so I suggest you not use klipper directly

wudixiaobinbin, 2021-08-06:
Ok, thank you, maybe waiting for klipper update