makerbase-mks/Klipper-for-MKS-Boards #10: Flashing Klipper firmware issue on Robin E3

omidta, 2021-09-21:
Please help.
I tried to flash my MKS Robin E3 by provided Robin_e3.bin but it sucked on 100%. I also made a new firmware by Klipper  based on the guide on generic-mks-robin-e3.cfg but no result. I disconnected my LCD but still the board didn't flashed

willngton, 2021-09-22:
> Please help.
> I tried to flash my MKS Robin E3 by provided Robin_e3.bin but it sucked on 100%. I also made a new firmware by Klipper based on the guide on generic-mks-robin-e3.cfg but no result. I disconnected my LCD but still the board didn't flashed

Hi @omidta 

This is normal, you just need to make sure file Robin_e3.bin on the SD-card changed to Robin_e3.CUR, and Klipper will be on the board MCU, go ahead and config you favorite web interface (octoprint, fluidd or mainsail)

Kind regards.

mks-viva, 2021-09-22:
1. You copy MKS Robin E3 path firmware bin file and update it
2. You need set you use the display screen
3. You need send 'firmware_restart' cammand

omidta, 2021-09-22:
Thank you @willngton  you saved a lot of my time

mks-viva, 2021-09-24:
So, I close this issues.

closed 2021-09-24 by mks-viva