makerbase-mks/Klipper-for-MKS-Boards #14: Publish instructions for Klipper compilation instead of providing pre-made binaries

VinnyCordeiro, 2021-11-04:
One of the advantages of Klipper is the ability to use multiple microcontroller boards. However, in order for that to work, all of them need to be compiled with the same version of Klipper. By providing pre-compiled binaries you are not allowing that to happen, because there isn't anyway to know which version of Klipper is being used. It would be much more useful to provide instructions on how to generate your own Klipper binary through the `make menuconfig` option.

mks-viva, 2021-11-05:
About how to compile the klipper, you can refer to Kilpper web, MKS path klipper bin file and it’s just used when people don’t want to compile klipper firmware.

mks-viva, 2021-11-05:
Klipper for MKS Board path, under each board folder, there are pictures about make_menuconfig

closed 2021-11-05 by VinnyCordeiro