makerbase-mks/Klipper-for-MKS-Boards #15: Klipper on a MKS Robin Nano v2

themacboy, 2021-12-03:
Good morning,

in your screen shoot of the menuconfig you suggest to select:

Serial (on USART3 PB11/PB10)

Are you sure? I can't connect with this parameters.

themacboy, 2021-12-03:
If I connect my TS35 screen I notice that the board is on "booting..."

Looks like the firmware is not working after the update. 

I have tryed with you precompiled bin and with my own compilation

themacboy, 2021-12-03:
@mks-viva any idea about this issue?

themacboy, 2021-12-03:
I answer my self.

Looking at klipper issues I have noticed that after make the firmware you must run the next script (to recode the output file):

./scripts/update_mks_robin.py ./out/klipper.bin ./out/Robin_nano35.bin

then you can flash the bin file and success with your configs.

 @mks-viva please add this in the readme file. Thx in advance !

willngton, 2021-12-04:
> I answer my self.
>
> Looking at klipper issues I have noticed that after make the firmware you must run the next script (to recode the output file):
>
> ./scripts/update_mks_robin.py ./out/klipper.bin [...]

If you have time, you can test TS35 connected to PI and use KlipperScreen
https://github.com/willngton/3DPrinterConfig/tree/main/mks_ts35

mks-viva, 2021-12-06:
@themacboy If you use board exp1,exp2 for MKS TS35 when you use klipper, MKS TS35 cann't work, because the klipper firmware have not support it.

themacboy, 2021-12-07:
> @themacboy If you use board exp1,exp2 for MKS TS35 when you use klipper, MKS TS35 cann't work, because the klipper firmware have not support it.
Thx, yes I know it.

The problem and solutionm was the one I have posted later.

bin file must be converted using the script:
./scripts/update_mks_robin.py ./out/klipper.bin ./out/Robin_nano35.bin

That was my real problem.

themacboy, 2021-12-07:
> If you have time, you can test TS35 connected to PI and use KlipperScreen https://github.com/willngton/3DPrinterConfig/tree/main/mks_ts35

Yes, I read it two days ago. I wil give a try when possible ! Thx

mks-viva, 2022-09-02:
MKS Pi + MKS PI-TS35 + your nano v2 and it's a good match. So, i will close it.

closed 2022-09-02 by mks-viva