makerbase-mks/Klipper-for-MKS-Boards #16: MKS Robin E3 blocked at end of flash

shawe, 2022-02-13:
Your provided bin for this board or the generated from klipper is causing to me that screen shows updated 100% and never reboot by itself.

The firmware is not working. I try to go back to Marlin and all seems fine then.

willngton, 2022-03-18:
> Your provided bin for this board or the generated from klipper is causing to me that screen shows updated 100% and never reboot by itself.
> 
> The firmware is not working. I try to go back to Marlin and all seems fine then.

I had similar issue, fixed re-creating the firmware but before make, run a make clean.

jdziadak, 2022-06-18:
I know that this issue is from February but maybe it will help someone.

First check your logs from klipper. Fluidd or Mainsail makes it pretty easy.

In my case from klipper log file I found out that it couldn't find serial that was defined in **printer.cfg**. 
In klipper repo at _klipper/config/generic-mks-robin-e3.cfg_ or _makerbase-mks/Klipper-for-MKS-Boards_ there is this line:
```
[mcu]
serial: /dev/serial/by-id/usb-1a86_USB2.0-Serial-if00-port0
```
I checked on my Thin Client running Raspberry Desktop in directory **/dev/serial/by-id/** and indeed I have an **usb-1a86_USB_Serial-if00-port0** instead **/dev/serial/by-id/usb-1a86_USB2.0-Serial-if00-port0** that is set in generic configs.
Simply changing it to serial name that is on my Thin Client resolved the issue.

Klipper will output to display (ofc. if its properly connected and configured in printer.cfg) if Host (pi/pc running klipper) will properly connect to serial from mcu on the printer (and probably no other issues are present). Otherwise your printer will stay at Updating 100% or BOOTING... (if you restarted your printer).

karelseaat, 2024-08-01:
so for a mks robin e2d v 1.1 i also did a update of the firmware.
it hangs at 100%, if i reboot it hangs at boot.
to build my firmwae i use these settings exactly: https://github.com/makerbase-mks/Klipper-for-MKS-Boards/blob/main/MKS%20Robin%20E3-E3D/Image/MKS%20Robin%20E3-E3D%20setting.png
i checked and double checked, ran a made clean rebuild it, same result.
at this point i doubt that a pi with klipper mainsail or anything will help.
Am i missing something ? is there an option to get some extra logging on the flashing sd card ? anything.

jdziadak, 2024-08-23:
> so for a mks robin e2d v 1.1 i also did a update of the firmware.
>
> it hangs at 100%, if i reboot it hangs at boot.
>
> to build my firmwae i use these settings exactly: [...]

I don't actively use 3d printer for about a year. 
As I understand you have flashed klipper onto board. Klipper is firmware and you need to interact with it. You can even do it via virtualization on Linux (virtualbox, etc.) and just connect to your pc. On this machine you need to install some form of front end like Mainsail or Fluidd. It won't boot without it as it makes mainboard a fancy gpio and every calculation is done on another machine (pi/vm/laptop).