makerbase-mks/Klipper-for-MKS-Boards #17: Robin nano v1.2 tft

manu7irl, 2022-03-05:
We need help to make it work on klipper, we need you to share some data about the TFT system in use there so we could, start developing for it, I got the flsun q5 works amazing in marlin touch ui, it should also work on klipper.

willngton, 2022-03-18:
> We need help to make it work on klipper, we need you to share some data about the TFT system in use there so we could, start developing for it, I got the flsun q5 works amazing in marlin touch ui, it should also work on klipper.

Robin Nano TFT35/24 is not suporte by Klipper not by MKS, 

You can have a look on a PR where someone manager to make similar tft works.
https://github.com/Klipper3d/klipper/pull/3724/

mks-viva, 2022-09-02:
https://es.aliexpress.com/item/3256804231556479.html?spm=a2g0o.store_pc_allProduct.8148356.7.20bd6688HGYwxl&pdp_npi=2%40dis%21USD%21US%20%2439.99%21US%20%2437.99%21%21%21%21%21%400bb47aa616620841942633473eba49%2112000029113611340%21sh

closed 2022-09-02 by mks-viva