makerbase-mks/Klipper-for-MKS-Boards #18: Klipper config for MKS PWC V3.0?

jonathanrobichaud4, 2022-03-10:
Does anyone have a Klipper config that I could use for the PWC V3.0?

viappidu, 2022-04-24:
Same issue here...

viappidu, 2022-04-26:
Did find a solution!
I connected the PW-OFF to the raspberry, +5v, GND and GPIO14 (they sit in a row, change as you like)
then in moonraker.conf I added:
```
[power Printer]
type: gpio
pin: gpio14
on_when_job_queued: True
off_when_shutdown: True
locked_while_printing: True
restart_klipper_when_powered: True
initial_state: off
```

This way I get a power printer button in moonsail interface. Pretty easy :)
If you like gcode you can also have a macro for M81 to switch it off
```
[gcode_macro M81]
gcode:
  {action_call_remote_method("set_device_power", device="Printer", state="off")}
```

---
I found also an idle timeout macro. Still working on it as does not behave as expected :( (not really spent much time on it)
```
[idle_timeout]
timeout: 180
gcode:
    { action_respond_info("Printer is idle") }
    TURN_OFF_HEATERS
    M84
    UPDATE_DELAYED_GCODE ID=auto_off DURATION=180 #wait this long before first periodic check
#   Idle time (in seconds) to wait before running the above G-Code
#   commands. The default is 600 seconds

[gcode_macro INTERRUPT_AUTO_SHUTDOWN]
gcode:
     UPDATE_DELAYED_GCODE ID=delayed_shutdown DURATION=0 #stop delayed_shutdown from getting executed
     { action_respond_info("Auto shutdown averted") }
     UPDATE_DELAYED_GCODE ID=auto_off DURATION=180 #start checking again

[delayed_gcode auto_off]
#This macro gets called delayed by the idle_timeout macro and checks for a few conditions before triggering a shutdown
#Conditions are: printer state is 'Idle', extruder temp is below or equal to 45°C, bed temp is below or equal to 60°C
initial_duration: 180
#klipper starts in the 'Idle' state but does not run the idle_timeout macro at startup,
#so there must be a non-zero value set here or the printer will not auto shutdown when
#it has not had another state than 'Idle' since the last klipper startup
gcode:
    { action_respond_info(printer.idle_timeout.state) }
    {% if printer.idle_timeout.state == "Idle" %} 
        {% if printer.extruder.temperature <= 45 and printer.heater_bed.temperature <= 60 %}
            { action_respond_info("Auto shutdown in 60 seconds, execute INTERRUPT_AUTO_SHUTDOWN to stop this.") } #warn user
            UPDATE_DELAYED_GCODE ID=delayed_shutdown DURATION=60
        {% else %}
            UPDATE_DELAYED_GCODE ID=auto_off DURATION=900 #printer was idle, but other conditions not met, check again later
        {% endif %}        
    #{% else %}
    #    { action_respond_info("Printer no longer idle") } #printer state was polled and no longer idle, stop repeating this routine. 
    {% endif %}

[delayed_gcode delayed_shutdown]
initial_duration: 0 
gcode:
    {% if printer.extruder.temperature <= 45 and printer.heater_bed.temperature <= 60 and printer.idle_timeout.state == "Idle" %}
        { action_respond_info("Shutting down now...") }
        { action_call_remote_method("set_device_power", device="Printer", state="off") } #CHANGE "PSU" TO YOUR DEVICE!
    {% else %}
        { action_respond_info("Auto shutdown averted") } #user did something to alter printer state
        UPDATE_DELAYED_GCODE ID=auto_off DURATION=900 #start checking again
    {% endif %}

```

jonathanrobichaud4, 2022-04-27:
Nice! However, i'm using octoprint but, I did find a solution that worked for me on this [reddit post](https://www.reddit.com/r/coreception/comments/n1z1ay/my_auto_powersleep_setup_with_klipper_on_the/). It's quite simple since the octoprint server handles all of the idle timeout stuff.

viappidu, 2022-04-27:
Yep,
I did see that, I started with that before going to moonraker solution.
Im using Octoprint too though I'm planning to leave it as soon as the 'delete objects' is fully implemented in klipper. Can't possibly think of not having it available for the moment...

ricardocolonia, 2022-09-02:
> Did find a solution! I connected the PW-OFF to the raspberry, +5v, GND and GPIO14 (they sit in a row, change as you like) then in moonraker.conf I added:
>
> ```
> [power Printer]
> type: gpio
> pin: gpio14 [...]

Would you explain a little bit better? 
Did you connected gpio 14 to the PWC V3.0 PW-OFF? or to the PW-OFF port on the board, what about the power button that comes with the PWC V3.0 did you plugged that one as well?

Thank you very much for your help

sima35, 2022-10-15:
> Našiel riešenie! Pripojil som PW-OFF k malinovke, +5v, GND a GPIO14 (sedia v rade, zmeňte ako chcete) potom v moonraker.conf som pridal:
> 
> ```
> [power Printer]
> type: gpio
> pin: gpio14
> on_when_job_queued: True
> off_when_shutdown: True
> locked_while_printing: True
> restart_klipper_when_powered: True
> initial_state: off
> ```
> 
> Týmto spôsobom dostanem tlačidlo napájania tlačiarne v rozhraní moonsail. Celkom jednoduché :) Ak máte radi gcode, môžete mať aj makro pre M81 na jeho vypnutie
> 
> ```
> [gcode_macro M81]
> gcode:
>   {action_call_remote_method("set_device_power", device="Printer", state="off")}
> ```
> 
> Našiel som aj makro pre časový limit nečinnosti. Stále sa na ňom pracuje, nespráva sa podľa očakávania :( (netrávil som na ňom veľa času)
> 
> ```
> [idle_timeout]
> timeout: 180
> gcode:
>     { action_respond_info("Printer is idle") }
>     TURN_OFF_HEATERS
>     M84
>     UPDATE_DELAYED_GCODE ID=auto_off DURATION=180 #wait this long before first periodic check
> #   Idle time (in seconds) to wait before running the above G-Code
> #   commands. The default is 600 seconds
> 
> [gcode_macro INTERRUPT_AUTO_SHUTDOWN]
> gcode:
>      UPDATE_DELAYED_GCODE ID=delayed_shutdown DURATION=0 #stop delayed_shutdown from getting executed
>      { action_respond_info("Auto shutdown averted") }
>      UPDATE_DELAYED_GCODE ID=auto_off DURATION=180 #start checking again
> 
> [delayed_gcode auto_off]
> #This macro gets called delayed by the idle_timeout macro and checks for a few conditions before triggering a shutdown
> #Conditions are: printer state is 'Idle', extruder temp is below or equal to 45°C, bed temp is below or equal to 60°C
> initial_duration: 180
> #klipper starts in the 'Idle' state but does not run the idle_timeout macro at startup,
> #so there must be a non-zero value set here or the printer will not auto shutdown when
> #it has not had another state than 'Idle' since the last klipper startup
> gcode:
>     { action_respond_info(printer.idle_timeout.state) }
>     {% if printer.idle_timeout.state == "Idle" %} 
>         {% if printer.extruder.temperature <= 45 and printer.heater_bed.temperature <= 60 %}
>             { action_respond_info("Auto shutdown in 60 seconds, execute INTERRUPT_AUTO_SHUTDOWN to stop this.") } #warn user
>             UPDATE_DELAYED_GCODE ID=delayed_shutdown DURATION=60
>         {% else %}
>             UPDATE_DELAYED_GCODE ID=auto_off DURATION=900 #printer was idle, but other conditions not met, check again later
>         {% endif %}        
>     #{% else %}
>     #    { action_respond_info("Printer no longer idle") } #printer state was polled and no longer idle, stop repeating this routine. 
>     {% endif %}
> 
> [delayed_gcode delayed_shutdown]
> initial_duration: 0 
> gcode:
>     {% if printer.extruder.temperature <= 45 and printer.heater_bed.temperature <= 60 and printer.idle_timeout.state == "Idle" %}
>         { action_respond_info("Shutting down now...") }
>         { action_call_remote_method("set_device_power", device="Printer", state="off") } #CHANGE "PSU" TO YOUR DEVICE!
>     {% else %}
>         { action_respond_info("Auto shutdown averted") } #user did something to alter printer state
>         UPDATE_DELAYED_GCODE ID=auto_off DURATION=900 #start checking again
>     {% endif %}
> ```

Good, could you send us a document for wiring, I don't want to burn anything?
Thank you in advance

sima35, 2022-10-15:
It worked correctly for me on how many GPIO15, GPIO14 did not communicate.
Is it a problem or not?

MagicalFire86, 2022-12-27:
MKS-PWC can be used with any board, it would be good to turn off PW-OFF after pressing it, so it turned off the system on the Raspberry Pi and then turned off the entire printer

marc05Arg, 2023-04-27:
> Even if you doesn't use a Raspberry Pi, you could still use MKS-PWC with a Robin Nano board (or, at least, with V3.x): [image] **CONNECTIONS:**
> 
> * Connect PW_DET to PA13 (it lets PWC detects the state of the board-mcu)
> * Connect PW-OFF to PB2 (so the board-mcu could send the signal to PWC to disconnect the power supply
> 
> Code (in printer.cfg, for example):
> 
> ```
> [output_pin PW_OFF]
> # Define output pin to power supply off
> pin: PB2
> value: 1 # Maintain power supply on
> 
> [gcode_macro PW_OFF_PRINTER]
> description: Shutdown 24V power supply
> gcode:
>   SET_PIN PIN=PW_OFF VALUE=0
> ```
> 
> In this mode, you could power off your printer (through power down the 24V power supply) in 3 ways:
> 
> 1. switching power switch to off
> 2. push on the button PW_OFF from the miscellaneous box (picture above)
> 3. push on the EMERGENCY STOP button
> 
> Of course, you could still use delayed and timeout macros adapted to this configuration (just using SET_PIN command).

Thanks CoolBeer, I wanted to know where PA13 is located on the board and if u can post a wiring diagram. 

Thks again.

Marcos.

ColdBeer72, 2023-04-27:
> Thanks CoolBeer, I wanted to know where PA13 is located on the board and if u can post a wiring diagram.
>
> Thks again.
>
> Marcos.

PA13 is the connector identified as PW_DET in your board (you could download the pin layout from the Makerbase's V3 GitHub)
I'm sorry I don't have a writing diagram, but it's just the usual connection diagram for this module.

marc05Arg, 2023-04-28:
> PA13 is the connector identified as PW_DET in your board (you could download the pin layout from the Makerbase's V3 GitHub) I'm sorry I don't have a writing diagram, but it's just the usual [...]

Gracias CoolBeer por la pronta respuesta, en mi placa RobinNano v1.2 PW_DET figura como PA2, de todas formas tengo conectados PW_OFF y PW_DET (Adjunto imagen) al PWC v3.0 y funciona correctamente con el firmware original, pero con Klipper se apaga. Gracias!

Thanks CoolBeer for the quick response, on my RobinNano v1.2 board PW_DET appears as PA2, anyway I have PW_OFF and PW_DET connected (image attached) to PWC v3.0 and it works correctly with the original firmware, but with Klipper it turns off. Thks!

[image]

marc05Arg, 2023-04-28:
I was able to **SOLVE IT!!!** with Kiauh and without physical modification. 

Running Klipper v0.11.0 on a RobinNano v1.2 board and MKS PWC 3.0 and Bltouch.

Greetings. Marcos.

[image]

[image]

gmaranda, 2023-06-18:
> I was able to **SOLVE IT!!!** with Kiauh and without physical modification.
> 
> Running Klipper v0.11.0 on a RobinNano v1.2 board and MKS PWC 3.0 and Bltouch.
> 
> Greetings. Marcos.
> 
> [image]
> 
> [image]

I found this reference trying to fix my issue.  Any chance you can state how you did it in Kiauh?  Thank you

marc05Arg, 2023-06-20:
> > I was able to **SOLVE IT!!!** with Kiauh and without physical modification.
> > Running Klipper v0.11.0 on a RobinNano v1.2 board and MKS PWC 3.0 and Bltouch.
> > Greetings. Marcos.
> > [image]
> > [image]
> 
> I found this reference trying to fix my issue. Any chance you can state how you did it in Kiauh? Thank you

HI gmaranda!!!   Of course, all you have to do is add these 2 GPIO parameters at the start of the microcontroller, compile and upload the firmware to the machine.

**PB2, !PA2**

I add an illustrative image and I hope it helps you.

Greetings. Marcos.

[image]

gmaranda, 2023-06-21:
> HI gmaranda!!! Of course, all you have to do is add these 2 GPIO parameters at the start of the microcontroller, compile and upload the firmware to the machine.
>
> **PB2, !PA2**
>
> I add an illustrative [...]

One last question did you modify the "printer.cfg"?

conting, 2023-06-21:
> Even if you doesn't use a Raspberry Pi, you could still use MKS-PWC with a Robin Nano board (or, at least, with V3.x): [image] **CONNECTIONS:**
>
> * Connect PW_DET to PA13 (it lets PWC detects the [...]

Hi CoolBeer,

if you add 
```
[gcode_button BUTTON_AUTOSHUTDOWN]
pin: ^!PA13
press_gcode: SET_PIN PIN=PW_OFF VALUE=0
```
the printer turn off when pressing the button, which comes with the pwc 3.0

So here is the complete code:
```
[output_pin PW_OFF]
# Define output pin to power supply off
pin: PB2
value: 1 # Maintain power supply on

[gcode_macro PW_OFF_PRINTER]
description: Shutdown 24V power supply
gcode:
  SET_PIN PIN=PW_OFF VALUE=0

[gcode_button BUTTON_AUTOSHUTDOWN]
pin: ^!PA13
press_gcode: SET_PIN PIN=PW_OFF VALUE=0
```

gmaranda, 2023-06-21:
> > Even if you doesn't use a Raspberry Pi, you could still use MKS-PWC with a Robin Nano board (or, at least, with V3.x): [image] **CONNECTIONS:**
> > 
> > * Connect PW_DET to PA13 (it lets PWC detects the state of the board-mcu)
> > * Connect PW-OFF to PB2 (so the board-mcu could send the signal to PWC to disconnect the power supply
> > 
> > Code (in printer.cfg, for example):
> > ```
> > [output_pin PW_OFF]
> > # Define output pin to power supply off
> > pin: PB2
> > value: 1 # Maintain power supply on
> > 
> > [gcode_macro PW_OFF_PRINTER]
> > description: Shutdown 24V power supply
> > gcode:
> >   SET_PIN PIN=PW_OFF VALUE=0
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > In this mode, you could power off your printer (through power down the 24V power supply) in 3 ways:
> > 
> > 1. switching power switch to off
> > 2. push on the button PW_OFF from the miscellaneous box (picture above)
> > 3. push on the EMERGENCY STOP button
> > 
> > Of course, you could still use delayed and timeout macros adapted to this configuration (just using SET_PIN command).
> 
> Hi ColdBeer,
> 
> if you add
> 
> ```
> [gcode_button BUTTON_AUTOSHUTDOWN]
> pin: ^!PA13
> press_gcode: SET_PIN PIN=PW_OFF VALUE=0
> ```
> 
> the printer turn off when pressing the button, which comes with the pwc 3.0
> 
> So here is the complete code:
> 
> ```
> [output_pin PW_OFF]
> # Define output pin to power supply off
> pin: PB2
> value: 1 # Maintain power supply on
> 
> [gcode_macro PW_OFF_PRINTER]
> description: Shutdown 24V power supply
> gcode:
>   SET_PIN PIN=PW_OFF VALUE=0
> 
> [gcode_button BUTTON_AUTOSHUTDOWN]
> pin: ^!PA13
> press_gcode: SET_PIN PIN=PW_OFF VALUE=0
> release_gcode: SET_PIN PIN=PW_OFF VALUE=1
> ```

Thank you for taking the time.  I will try it.

ColdBeer72, 2023-06-22:
Thanks a lot for completing the circle, @conting 

Now, it's time Makerbase take note of how to give complete functionality to the PWC AGAIN, and give these instructions in their git.

Let's see if they read here.

ColdBeer72, 2023-06-22:
So, thanks to @gmaranda , we have got a running PWC configuration in klipper. To add more usability and manage it from mainsail and moonraker, I'll change the code:

In printer.cfg:
```
[output_pin PW_OFF]
# Define output pin to power supply off
pin: PB2
value: 1 # Maintain power supply on

[gcode_macro PW_OFF_PRINTER]
description: Shutdown 24V power supply
gcode:
  SET_PIN PIN=PW_OFF VALUE=0

[gcode_button BUTTON_AUTOSHUTDOWN]
pin: ^!PA13
press_gcode: SET_PIN PIN=PW_OFF VALUE=0
release_gcode: SET_PIN PIN=PW_OFF VALUE=1
```

In moonraker.conf:
```
# Control a relay providing power to the printer
[power printer]
type: klipper_device
object_name: output_pin PW_OFF
```
PD: This is a work-in-progress, as I don't have PWC connected now (I finished using a Tasmota plug to manage the printer power). Please, test it and let comments here.

gmaranda, 2023-06-22:
> So, thanks to @gmaranda , we have got a running PWC configuration in klipper. To add more usability and manage it from mainsail and moonraker, I'll change the code:
>
> In printer.cfg:
>
> ```
> [output_pin [...]

Hi!

Altough this works it is not fully there yet.  When I start my machine I have to hold  the button for about 30sec or until you hear the click inside the PWC.  After that the machine stays on and if you don't wait long enough the machine shuts down.  When the machine is running you get the shutdown button in mainsail (for me) and if you push it this will shut down the machine.  Not too bad for now but at least the machine works.

I tried to add the PB2, !PA2 as suggested by Marcos (Marc05Arg) above but it would not load properly.  At least this is progress but the work continues.

ColdBeer72, 2023-06-23:
You're right, @gmaranda ... I would love if @mks-viva or @makerbase-mks follow our work and give lasts steps.

conting, 2023-06-23:
Hi @ColdBeer72 @gmaranda,

as a workaround for this Problem I added this to the printer.cfg file:

```
[delayed_gcode PWC_ON]
initial_duration: 1
gcode:
  SET_PIN PIN=PW_OFF VALUE=1
```

So now it looks like this:

```
[delayed_gcode PWC_ON]
initial_duration: 1
gcode:
  SET_PIN PIN=PW_OFF VALUE=1

[output_pin PW_OFF]
# Define output pin to power supply off
pin: PB2
value: 1 # Maintain power supply on

[gcode_macro PW_OFF_PRINTER]
description: Shutdown 24V power supply
gcode:
  SET_PIN PIN=PW_OFF VALUE=0

[gcode_button BUTTON_AUTOSHUTDOWN]
pin: ^!PA13
press_gcode: SET_PIN PIN=PW_OFF VALUE=0
release_gcode: SET_PIN PIN=PW_OFF VALUE=1
```

After turning on the printer I hear a "click click" and the printer stays on. I know this is just a workaround, buts its better than nothing. If I find a better solution I will post it here.

gmaranda, 2023-06-23:
> Hi @ColdBeer72 @gmaranda,
>
> as a workaround for this Problem I added this to the printer.cfg file:
>
> ```
> [delayed_gcode PWC_ON]
> initial_duration: 1
> gcode:
>   SET_PIN PIN=PW_OFF VALUE=1
> ```
>
> So now it [...]

Thank you for the idea.  Unfortunately for me it does not work.  I have put  your codes as is and the machine boots etc. but I still have to hold the button and count to about 30 for the machine to stay on.  I must be cursed!  I left it in the config file since it does not seem to do anything wrong.  I am glad it is working for you though.  thanks again

J0k3R92, 2023-08-24:
First off, thank you to everyone that's been collectively worked on this project.  Below is a quick reference of how I've gotten the MKS PWC V 3.0 to work on a setup using the stock MCU and no Pi.

Printer:  FLSUN Q5
MCU:  Robin Nano v1.2

MKS PWC Wiring:
    PW-DET [PWC] connected to PA2 [Robin Nano]
    PW-OFF [PWC] connected to PB2 [Robin Nano]

Klipper Firmware Build:
    Set PA2 and PB2 GPIO pins on startup 

[image]

Addition to printer.cfg
    [output_pin PW_OFF]
    # Define output pin to power supply off
    pin: PB2
    value: 1 # Maintain power supply on

    #[gcode_macro PW_OFF_PRINTER]
    ## enables control of a relay providing 24V power to the printer via a macro button on the mainsail DashBoard
    #gcode:
    #    SET_PIN PIN=PW_OFF VALUE=0

    #[gcode_button BUTTON_AUTOSHUTDOWN]
    #pin: ^!PA2
    #press_gcode: SET_PIN PIN=PW_OFF VALUE=0
    #release_gcode: SET_PIN PIN=PW_OFF VALUE=1

Addition to moonraker.cfg
    # enables control of a relay providing 24V power to the printer via a button under the miscellaneous section of the mainsail DashBoard
    [power printer]
        type: klipper_device
        object_name: output_pin PW_OFF

I chose to keep [gcode_macro PW_OFF_PRINTER] disabled is it's a duplicate of the button configured in the moonraker.cfg and only one or the other is really needed.  However, it's still useful for those who would like to append the macro to the end of their GCODE files to shutdown after the completion of a print though.

I also chose to keep [gcode_button BUTTON_AUTOSHUTDOWN] disabled as I've found a purpose for not allowing it to act a a reset switch.  The current config [with no use of a raspberry Pi and only having the onboard MCU setting the PWC] causes the printer to shutdown and turn off anytime a firmware reset command is given.  However, if you hold the reset button down while performing the firmware reset, the printer stays on throughout the cycle.  So enabling/disabling this portion has it's own benefits.

The below screenshot shows the two sections of the DashBoard described above.

[image]

STA4RD, 2025-07-11:
Thank you all for your work — it works perfectly.
I’d like to add something:

I power the host (Orange Pi Zero 3) from a separate 5V power supply. The MCU is connected via UART, and it’s enough to feed 5V from the host to the USB-PWR ON pin (or J17 3 pin) to make everything come back online after a reboot without having to hold the button.
I believe that if the MCU is connected via USB instead, it should be enough to move the jumper on USB-PWR from OFF to ON. That should achieve the same result.

[image]