makerbase-mks/Klipper-for-MKS-Boards #24: Compiling Klipper for RobinE3 Result

dlhenke, 2022-11-13:
I'm trying to compile new firmware for Robin_e3 all compilation goes well but the result is unusable, since klipper does not connect to the board anymore. 
my printer.cfg file is identical , also the result of ls /dev/serial/by-id/* is the same
/dev/serial/by-id/usb-1a86_USB_Serial-if00-port0
My board version is Robin e3 v.1.1
My problem with the stock klipper firmware is Bltouch behaviour that  was working well with marlin ..
[image]
[image]

dlhenke, 2022-11-13:
My mistake, i Forget to execute the script ./scripts/update_mks_robin.py out/klipper.bin out/Robin_e3.bin
also in printer.cfg must be 
[bltouch]
#sensor_pin: ^PC6 
sensor_pin: ^PB1
control_pin: PA3
[image]

nuqz, 2023-01-10:
@dlhenke Hi! May I ask you to share your config (the results of `make menuconfig`) file for this board, please?

ColdBeer72, 2023-01-10:
> @dlhenke Hi! May I ask you to share your config (the results of `make menuconfig`) file for this board, please?

Hi @nuqz , your answer is well explained in three points:

1. The screenshot from @dlhenke in the first post
2. The **[screenshot](https://github.com/makerbase-mks/Klipper-for-MKS-Boards/blob/main/MKS%20Robin%20E3-E3D/Image/MKS%20Robin%20E3-E3D%20setting.png)** shared by Makerbase for your printer 
3. And the **[cfg](https://github.com/makerbase-mks/Klipper-for-MKS-Boards/blob/main/MKS%20Robin%20E3-E3D/generic-mks-robin-e3-v1_1.cfg)** shared in this github for your board which says at the start:

```
# This file contains common pin mappings for MKS Robin E3 boards. To
# use this config, the firmware should be compiled for the STM32F103.
# When running "make menuconfig", enable "extra low-level
# configuration setup", select the 20KiB bootloader, and serial (on
# USART1 PA10/PA9) communication.

# Note that the "make flash" command does not work with MKS Robin
# boards. After running "make", run the following command:
#   ./scripts/update_mks_robin.py out/klipper.bin out/Robin_e3.bin
# Copy the file out/Robin_e3.bin to an SD card and then restart the
# printer with that SD card.

# MKS Robin E3 has onboard TMC2209. This config can also be used for
# MKS Robin E3D if equipped with TMC2209 stepper drivers.
```

nuqz, 2023-01-10:
@ColdBeer72 Thanks a lot for your help! Now it's clear to me.