makerbase-mks/Klipper-for-MKS-Boards #27: Klipper v0.11.0 firmware update

Neko-vecter-Lab, 2022-12-05:
Hey! Klipper now version `v0.11.0`. The firmware in this repo is outdated.
It is possible to update the firmware in this repo?

ColdBeer72, 2023-01-10:
@nuqz I explained this for you in other issue #24

Neko-vecter-Lab, 2023-01-10:
> I join the question. It is difficult for me to understand the build of the firmware on my own, or rather, I am afraid to miss the configuration of some feature (e.g. GPIO pins state on startup) of my `robin e3` and burn the board. Could you, please, at least add the configurations (the results of `make config`) that were used for the previous `0.10.x` version to the repository so that we have a good starting point?
> 
> UPD. Do I understand correctly that I only need to configure as shown on [this image](https://github.com/makerbase-mks/Klipper-for-MKS-Boards/blob/main/MKS%20Robin%20E3-E3D/Image/MKS%20Robin%20E3-E3D%20setting.png) and everything will be fine?

The reason why I post this issue is because klipper update to the v0.11.0. The repo firmware is outdated. 

Build the klipper firmware is not hard. You just need to follow the klipper docs. 
The compile parameters have been posted in this repo. If you want the specific version of firmware, you need to compile.