makerbase-mks/Klipper-for-MKS-Boards #39: robin nano v1.2 not booting

Mudardepois, 2023-09-07:
When the TFT updating screen reaches 100%, it freezes. After a restart, the screen only displays "booting," and nothing else occurs.

jpokorny12, 2023-09-21:
Hi, i have same problem. I use this setting for build firmware [https://stone-time.de/docs/allgemeine-anleitung-zum-erzeugen-einer-klipper-firmware/](url), but it didnt work. Now i try precompiled fw from this github, and it works ;)

JasonChio365, 2024-01-19:
> When the TFT updating screen reaches 100%, it freezes. After a restart, the screen only displays "booting," and nothing else occurs.

Hello, I have also encountered the problem that the touch screen is frozen at 100%, do you have any solution?
[image]

ycq3, 2024-02-29:
1. 需要执行脚本转换固件`./scripts/update_mks_robin.py out/klipper.bin out/Robin_nano35.bin`
2. 需要配置`GPIO pins to set at micro-controllerstartup` 值为 `!PC6,!PD13`
我配置后成功启动，但是屏幕没有显示，我还在解决屏幕的显示问题。

ColdBeer72, 2024-02-29:
> 1. Need to execute script to convert firmware`./scripts/update_mks_robin.py out/klipper.bin out/Robin_nano35.bin`
> 2. `GPIO pins to set at micro-controllerstartup`The value needs to be configured. `!PC6,!PD13`
>    After I configured it, it started successfully, but the screen did not display. I am still solving the screen display problem.

The screen will NEVER work. In the same way. There are some workarounds to make it work but not connected to three MKS Robin Nano board.
I don't know if you your point 2 is specific for a locked board, but it's definitely not necessary for a Klipper first install nor an update, on a working board..

fred794, 2024-06-25:
Version: v0.12.0-166-g957475421
Would it be possible to compile in this version because it doesn't work for me?

closed 2024-08-05 by Mudardepois